datacute/Tiny4kOLED #1: Extra dots

entozoon, 2018-03-23:
Hey dude, this looks like a great improvement on the digisparkOled but I'm having a little problem -
Running the example (with no modifications) draws extra pixels. Hard to see exactly what's happening but it possibly only occurs during fill and clear (not print).
A very consistent grid of problem pixels, perhaps it's chopping off a bit of a byte somewhere?

It doesn't occur with the digispark version but I'd rather not use that as it's slow AF

[image]

datacute, 2018-03-23:
Yeah, sorry, I know exactly what the problem is... Adafruit haven't created a new release of TinyWireM containing the bug fix that stops the buffer overrun. Are you happy to edit your downloaded copy of their library?

datacute, 2018-03-23:
https://github.com/adafruit/TinyWireM/pull/6/commits/b5698bf38df0e81c72c7ac5a0ba43cd88314ee77

entozoon, 2018-03-23:
Oh wow, prompt response thank you!
Ah, I see you've repaired TinyWireM, nice job. I'll give it a go tonight 👍

entozoon, 2018-03-23:
That totally worked, I'm chuffed. Thanks mate :)
Now to see if I can bodge it to load a 32px tall font...

datacute, 2018-04-02:
I'd prefer to leave this issue open, waiting for an Adafruit release of TinyWireM, or a workaround in this library. I'm also thinking about comparing SpenceKonde/ATTinyCore wire library.