datacute/Tiny4kOLED #7: Define display type?

classic-audio, 2018-07-14:
How do I define when I connect:
OLED 64x128 display ?
OLED 32x128 display?
(driver SSD1306)

datacute, 2018-07-14:
This library is targeting 128x32 displays. You can supply a different init sequence to the begin call, but the text lines and double buffering are coded for 32 lines. I'm thinking of cloning the library for 64x48 and for 128x64.

classic-audio, 2018-07-14:
Perhaps this is only me, but several of the files refer to the 128x64 display.
"Drivers for SSD1306 controlled dot matrix OLED/PLED 128x64 displays"
Of course I have seen the explain of dobb. buffering also.

A very positive point.
Your Tiny4kOLED and TinyWire - is so far the only libraries I can find which can work with ATtiny85 and 2xINA219 (current / voltage measurement). 
The resorces of the ATtiny85 is only 85% and it works great - except for the 128x64 OLED of course  

When do you plan to have the solution for 128x64 OLED?

datacute, 2018-07-14:
Thanks for drawing my attention to that, I'll see where I can clarify the descriptions.

Unfortunately I don't have a 128x64 OLED using the SSD1306 to test with.

The double buffering technique won't work, as all the SSD1306's ram is displayed for a 128x64 screen (except perhaps by using double height lines in zoom mode, depending on the hardware configuration).

So assuming you don't use the double buffering calls, you can alter the library code, there's at least one line of code to change:
#define SSD1306_PAGES 4
Needs to change to 
#define SSD1306_PAGES 8
That change is needed so that the clear() command clears the entire screen instead of just half of it.

You also need to determine the initialisation commands to set up the hardware. I commented out a lot, as they were setting the value that the chip has at reset anyway, and the saving of a few bytes can be important. Simple commenting out the setting of the multiplexor (that's the 0xA8, 0x1F line), might be enough as it defaults to 0x3F, but you might have to change other settings too, such as the com pins hardware configuration, and COM output scan direction.

The SSD1306 supports many different ways of connecting the screen to the driver chip, which is why I added the override of the begin() method so that different hardware could be configured in different ways. If you have another library that works with your screen, have a look in its source code, trying to find the sequence of command bytes it sends when it initialises the display.

The code before I changed it to be for a 128x32 display, used to initialise using this sequence:
    static const uint8_t ssd1306_init_sequence [] PROGMEM = {	// Initialization Sequence
    	0xAE,			// Display OFF (sleep mode)
    	0x20, 0b00,		// Set Memory Addressing Mode
    					// 00=Horizontal Addressing Mode; 01=Vertical Addressing Mode;
    					// 10=Page Addressing Mode (RESET); 11=Invalid
    	0xB0,			// Set Page Start Address for Page Addressing Mode, 0-7
    	0xC8,			// Set COM Output Scan Direction
    	0x00,			// ---set low column address
    	0x10,			// ---set high column address
    	0x40,			// --set start line address
    	0x81, 0x3F,		// Set contrast control register
    	0xA1,			// Set Segment Re-map. A0=address mapped; A1=address 127 mapped. 
    	0xA6,			// Set display mode. A6=Normal; A7=Inverse
    	0xA8, 0x3F,		// Set multiplex ratio(1 to 64)
    	0xA4,			// Output RAM to Display
    					// 0xA4=Output follows RAM content; 0xA5,Output ignores RAM content
    	0xD3, 0x00,		// Set display offset. 00 = no offset
    	0xD5,			// --set display clock divide ratio/oscillator frequency
    	0xF0,			// --set divide ratio
    	0xD9, 0x22,		// Set pre-charge period
    	0xDA, 0x12,		// Set com pins hardware configuration		
    	0xDB,			// --set vcomh
    	0x20,			// 0x20,0.77xVcc
    	0x8D, 0x14,		// Set DC-DC enable
    	0xAF			// Display ON in normal mode
    	
    };

classic-audio, 2018-07-14:
Thank you very much.
I will try your suggestions, for me this is a complicated task. I hope I can do it by following your guidelines.

IG-77, 2019-05-08:
Has this progressed any further?

closed 2019-05-09 by classic-audio

GiorgosXou, 2020-01-19:
**Found out a library** ["here" ](https://www.elektormagazine.com/labs/oled-usb-voltage-and-current-meter-updated), that i've managed to use with success _(on an attiny85 with ssd1306 128x**64**)_ **without any issues** _(it could help with the issue, maybe... or at least someone who is searching for a solution in this topic])_.

_have a nice day people,  GN 😴💛_

**PS**. Although i first Tried to find a "fix" [...]

datacute, 2020-02-07:
I've added support for setting the number of pages to clear.