datacute/Tiny4kOLED #14: ATTINY84 crashes

flyandi, 2019-06-02:
It seems this library was designed for the 85 series but other than pin layout it seems there are no real differences - so I assume it should run fine on the 84. Right now the lib crashes and the 85 and the Tiny is sent to an indefinite loop :(

Does anyone has successfully run this on an 84?

Thanks,

Andy

datacute, 2019-09-10:
Sorry for taking so long to respond.
I'd like to help, but don't have an ATtiny84 to test with.
I'm not aware of any code that is specific to the 85.
Do you get the same amount of dynamic memory left for local variables when you compile your project? I've had programs freeze due to running out of ram.
Which I2C implementation are you using? Are you in a position where you can try others?
Are you using SpenceKonde/ATTinyCore?

alrabe, 2020-05-31:
Hi. I have running this on a 84 with an 128x64 oled. The program will freeze after some time but I didn't check yet why this happens. It is more likely caused by my own code.

SpenceKonde, 2023-02-27:
It sounds like this bug, if it was ever a bug in the library and not user code, has been long since solved, as no comments in 3 years. Probably closable