datacute/Tiny4kOLED #16: Using scroll commands

elgranshaman, 2019-08-31:
Hello,

Would it be possible to add an example about how to use the scrolling commands?

`		void scrollRight(uint8_t startPage, uint8_t interval, uint8_t endPage);
		void scrollLeft(uint8_t startPage, uint8_t interval, uint8_t endPage);
		void scrollRightOffset(uint8_t startPage, uint8_t interval, uint8_t endPage, uint8_t offset);
		void scrollLeftOffset(uint8_t startPage, uint8_t interval, uint8_t endPage, uint8_t offset);
		void deactivateScroll(void);
		void activateScroll(void);
		void setVerticalScrollArea(uint8_t top, uint8_t rows);
`

Not sure how to use these, thank you.

datacute, 2019-09-01:
I don't think that the scroll functions are terribly useful, due to there being no frame synchronisation possible. I wrote a blog post about that here: http://www.datacute.nz/2018/09/28/ssd1306-fr-output.html

I've used the scroll functions in a title screen for a game, where six lines of text instructions scroll through with about two and a half lines of text showing on the screen, keeping the game title stationary at the top, and with an additional small horizontal scroll of a divider at the end of the instructions.
The code is here:
https://github.com/datacute/TinyBullsAndCows/blob/master/TinyBullsAndCows.ino#L128

datacute, 2020-02-07:
I've copied the relevant code into a new example available in version 1.3.0 of this library.

datacute, 2020-03-07:
I'm re-opening this issue, since I've just discovered that the **SSD1306B** chip used in most of these screens can be asked to scroll a portion of the screen horizontally by 1 column!

All the white screens I have do this, shifting in zeroes.
The blue screen I have does this too, but appears to wrap the data instead of shifting in zeroes.

I'm adding support and examples to the library.

(Search the internet for SSD1306B to find the datasheet and look for the 2C/2D "Content Scroll
Setup" command.)

datacute, 2020-03-28:
I added a new method to draw a portion of a text string, starting from a certain number of pixels in to the string, and drawing a limited width in pixels. This allows for both clipping text, and filling in one column of pixels to enable scrolling.
A ScrollingText example has been added in v1.4.1

closed 2020-03-28 by datacute