datacute/Tiny4kOLED #17: Adding 16x32 Fonts

elgranshaman, 2019-08-31:
Hello,

I have found on [this website](https://sites.google.com/site/pre2arduino/home/huge-font-on-a-tinny-oled-atiny85-1) a modification of this library to support 16x32 Fonts. I can see that the code is quite different so I am not sure if this can be integrated straight away or it was done for an earlier release.

It would be nice to have this on the official library and to have the option of not uploading into program memory fonts that are not being used with the idea of saving program space.

Thank you

datacute, 2019-09-01:
Hi,
I had seen that project.
I think it does require modification to work with my extended font support, in that I think the data was all needing to be rotated or flipped x-y. As it is presented in that website, the source code does have a nice feature that the font data represented in binary makes the font characters appear. (As shown in the comments.) I think I went in a different direction trying to get the most speed for the least amount of code instead.
My font support only loads data for fonts that are used. I've added some more fonts to a new library https://github.com/datacute/TinyOLED-Fonts
I will look at making that font available there too.

elgranshaman, 2019-09-02:
Thanks a lot. Yes, I will definitely be good to have the option of a large font.

Thank you

datacute, 2019-09-07:
I've converted and added that font to https://github.com/datacute/TinyOLED-Fonts

closed 2019-09-07 by datacute