datacute/Tiny4kOLED #21: external crystal problem

classic-audio, 2019-11-30:
I have a sketch working fine on ATtiny85 with internal 8MHz setup.
I have then used "burn bootloader" external 8MHz - uploaded the code again (setup with the 8MHz crystal).
But after succesful code loading - the ATtiny85 dont work with the 128x32 OLED display.
Why?

classic-audio, 2019-11-30:
I have loaded the LED / blink sketch to **ATtiny85 ext. 8MHz crystal** and thank works fine.
Then I added one tekst line and the Tiny4kOLED.h to the LED / blink sketch.
The LED blink function work, but no tekst on the OLED

datacute, 2019-12-11:
Sorry for taking a while to respond. I've never used the External Crystal clock options on the ATTiny85. I've purchased some cheap crystals, including 8MHz, and I'm waiting for them to arrive so I can experiment. Are you using Spence Konde's ATTinyCore?

classic-audio, 2019-12-12:
Thank you - I hope for a good solution.
Are you using Spence Konde's ATTinyCore? - no.

classic-audio, 2020-02-09:
I can now use an external xtal with the ATtiny85.
(the library update jan 2020)

closed 2020-02-09 by classic-audio

datacute, 2020-02-15:
I've finally received done crystals, and tested, and confirm what you saw. The change made in January to use a more robust OLED initialisation fixes the problem. I also found that resetting the ATtiny85 would work, as would waiting longer before initialising the OLED. It's the same behavior as seen with the blue OLED, which is what prompted me to investigate the startup code in January. Thanks for reporting the issue.