datacute/Tiny4kOLED #22: OLED 128x32 - rotate 180dg?

classic-audio, 2020-02-09:
I have been seaching for a function to rotate the text 180dg on a OLED 128x32 display, but I cannot find such a command.
Is it possible to rotate the text on the 128x32 OLED display?

datacute, 2020-02-14:
Thanks for asking. Yes, it is possible to rotate the entire display 180 degrees.

There are two settings (setComOutputDirection and setSegmentRemap), the result in mirroring in either axis, and in the latest version I made this even easier, and more obvious, by adding a `oled.setRotation(0)` method.

I'll add that method in to the examples.

There isn't a way of rotating *just* the text, but I think that the text is the only thing where it matters what the orientation is, so rotating the entire display should work.

closed 2020-02-14 by datacute

mohitbhoite, 2022-12-25:
Thank you for the awesome library! Is there a way to rotate the display (ssd1306 128x32) by 90 degrees?

datacute, 2022-12-26:
Sorry @mohitbhoite no there isn't. The library makes extensive use of the ssd1306 memory structure being 'pages' of 8 pixels mapping to the display in various ways, but not rotated 90 degrees.