datacute/Tiny4kOLED #24: Compile error "void clipText(uint16_t startPixel, uint8_t width, const __FlashStringHelper *text);"

GFA-SkyRanger, 2020-04-17:
Arduino: 1.8.12 (Windows 7), Board: "Digispark (Default - 16.5mhz)"
```

In file included from E:\SkyRanger\Eigene Dateien\Arduino\libraries\Tiny4kOLED-1.5.0\src/Tiny4kOLED_Wire.h:12:0,

                 from E:\SkyRanger\Eigene Dateien\Arduino\libraries\Tiny4kOLED-1.5.0\src/Tiny4kOLED.h:19,

                 from R:\Arduino\DigisparkOLED_\DigisparkOLED_.ino:2:

E:\SkyRanger\Eigene Dateien\Arduino\libraries\Tiny4kOLED-1.5.0\src/Tiny4kOLED_common.h:74:59: error: '__FlashStringHelper' does not name a type

   void clipText(uint16_t startPixel, uint8_t width, const __FlashStringHelper *text);

                                                           ^

E:\SkyRanger\Eigene Dateien\Arduino\libraries\Tiny4kOLED-1.5.0\src/Tiny4kOLED_common.h:74:80: error: ISO C++ forbids declaration of 'text' with no type [-fpermissive]

   void clipText(uint16_t startPixel, uint8_t width, const __FlashStringHelper *text);

                                                                                ^

```

datacute, 2020-04-17:
Oh that's annoying. Thanks for reporting it.

I was wondering when I added that, how to do it in a way that would work on all boards. I use https://github.com/SpenceKonde/ATTinyCore 

Could you tell me what you've got for your "Additional Board Manager URLs" preferences setting, to help me figure out a solution?

GFA-SkyRanger, 2020-04-17:
Hi,
i am using the following Link for digispark-boards:
http://digistump.com/package_digistump_index.json

datacute, 2020-04-17:
Thanks.

The intention of the code is to allow the equivalent of the F("string") macro to put the text into PROGMEM, but also knowing the length of the text.
This can be seen in the ScrollingText example:

https://github.com/datacute/Tiny4kOLED/blob/master/examples/ScrollingText/ScrollingText.ino#L32

```
const char textToScrollData[] PROGMEM = { "This is an example of scrolling text" };
const __FlashStringHelper * textToScroll = (__FlashStringHelper*)textToScrollData;

```
https://github.com/datacute/Tiny4kOLED/blob/master/examples/ScrollingText/ScrollingText.ino#L94

`if (nextRowOfTextToDraw >= sizeof(textToScrollData) * 8) {`

GFA-SkyRanger, 2020-04-17:
Well, i didn't try it with other boards like the "normal" arduino boards.
I just would a solution to display bigger digits.

i later found an "tiny8k-oled" library, which includes the big font 16x32, which is perfect for my project.
Of course, it was VERY big for the small attiny85, but with some rewriging of the library i could spare some kilobits.

This board i used for the Project:
https://gfaclan.de/cloud/index.php/s/DoDdyYX8tdSGReq

This is the working one:
https://gfaclan.de/cloud/index.php/s/rr9rYXb9LGQp7KH

datacute, 2020-04-17:
Here's someone else with the same problem:
https://forum.arduino.cc/index.php?topic=397627.0

v1.4.0 of my library should work for you while I try and find the best solution.
There is a 6x32 digit font in https://github.com/datacute/TinyOLED-Fonts

datacute, 2020-04-17:
Is that the "Raspberry Pi" DS3231 RTC board? I've been thinking about making a little clock too, using A DS3231 board of some kind.

GFA-SkyRanger, 2020-04-18:
Yupp, its the board for Pi.
It was never planned to get the modules special for the PI. In principle it doesn't matter what the board looks like, the main thing is that the RTC chip is supported by a library.

datacute, 2020-04-19:
I'm having trouble compiling anything with the digistump boards...

`exec: "C:\\Users\\Stephen\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\avr-gcc\\4.8.1-arduino5/bin/avr-g++": file does not exist`

So are you able to try changing Tiny4kOLED_common.h to include the below code within it?

```
#ifndef F
class __FlashStringHelper;
#define F(string_literal) (reinterpret_cast<const __FlashStringHelper *>(PSTR(string_literal)))
#endif
```

GFA-SkyRanger, 2020-04-19:
then i get this error:

`In file included from E:\SkyRanger\Eigene Dateien\Arduino\libraries\Tiny4kOLED-1.5.0\src/Tiny4kOLED_Wire.h:12:0,

                 from E:\SkyRanger\Eigene Dateien\Arduino\libraries\Tiny4kOLED-1.5.0\src/Tiny4kOLED.h:19,

                 from E:\SkyRanger\Eigene Dateien\Arduino\DigiSpark_Uhr\DigiSpark_Uhr\DigiSpark_Uhr.ino:2:

E:\SkyRanger\Eigene Dateien\Arduino\libraries\Tiny4kOLED-1.5.0\src/Tiny4kOLED_common.h:79:59: error: '__FlashStringHelper' does not name a type

   void clipText(uint16_t startPixel, uint8_t width, const __FlashStringHelper *text);

                                                           ^

E:\SkyRanger\Eigene Dateien\Arduino\libraries\Tiny4kOLED-1.5.0\src/Tiny4kOLED_common.h:79:80: error: ISO C++ forbids declaration of 'text' with no type [-fpermissive]

   void clipText(uint16_t startPixel, uint8_t width, const __FlashStringHelper *text);

                                                                                ^`

datacute, 2020-04-19:
Thanks for trying. I have one more idea for a simple fix - can you try adding just the middle two lines?

```
class __FlashStringHelper;
#define F(string_literal) (reinterpret_cast<const __FlashStringHelper *>(PSTR(string_literal)))
```

GFA-SkyRanger, 2020-04-19:
Now it works but the following code wont work anymore:
`oled.print(F("0"));`

datacute, 2020-04-19:
Ah, ok - I've found digistump's F macro definition.
You can use just the one line:

```
class __FlashStringHelper;
```

That should fix print, but you probably can't call the new `clipText` method like `oled.clipText(0,10,F("example"));` (which I was hoping to allow for) but only like how I do in the ScrollingText example.

That `__FlashStringHelper` is just a marker class, only ever used to identify a pointer as being to a PROGMEM string. I think I can figure out now how to make it work as I intend for both the digistump core and Spence Konde's ATTinyCore.

GFA-SkyRanger, 2020-04-19:
Yupp, it works.

datacute, 2020-04-20:
I'm considering replacing `const __FlashStringHelper` in Tiny4kOLED_common.h and Tiny4kOLED.cpp with `DATACUTE_F_MACRO_T` defined in Tiny4kOLED_common.h as

```
#ifndef DATACUTE_F_MACRO_T
#ifdef AVR_DIGISPARK
#define DATACUTE_F_MACRO_T fstr_t
#else
#define DATACUTE_F_MACRO_T const __FlashStringHelper
#endif
#endif
```

datacute, 2020-04-22:
Fixed in release 1.5.1

TobyBorland, 2020-05-17:
Compiles for Digistump non-pro with,

#ifndef AVR_DIGISPARK

in Tiny4kOLED_common.h

datacute, 2020-05-18:
Yeah - sorry I hadn't quite figured out that the various entries in digispark's ```boards.txt``` like
```digispark-tiny.build.board=AVR_DIGISPARK```
ended up having ```ARDUINO_``` put on the front by the ```recipe.c.o.pattern``` containing
```-DARDUINO_{build.board}``` in the ```platform.txt```.
I'll correct the library, and release another version.

TobyBorland, 2020-05-22:
Compiles for Mellis library as #ifdef AVR_DIGISPARK
Tested on a naked ATtiny85, upload via Arduino as ISP using modified 32x16 fonts.

[image]

datacute, 2020-05-23:
I've just released v1.5.2. I'm looking for confirmation that it compiles without modifications using digispark/digistump (and/or David Mellis's) board manager implementations.
Note: I think Digispark's Wire.h is a rebadged TinyWireM, without the buffer overrun fix that is in Adafruit's TinyWireM.

gitterman, 2020-07-13:
I ran into the same problem with my Digispark and lib 1.5.1 which I had installed using the library manager of the 1.8.12 IDE.
I am glad I found the solution to the issue here as the error message of the IDE was somehow confusing to me.
I downloaded the 1.5.2 and copied it over the 1.5.1 in my lib location and now the examples work like a charm.

datacute, 2020-07-14:
Thanks for the confirmation @gitterman. Was the 1.5.2 version not showing up as an update in the library manager? (It might be hard to tell now, if you hadn't checked before finding this issue and downloading it.)

closed 2020-07-14 by datacute

gitterman, 2020-07-14:
the 1.5.2 was not showing up and is still not showing.
As I said I just copied the files from github over the existing 1.5.1.
The library manager still shows 1.5.1 as the installed version and does not offer an update or a 1.5.2 version as shown on the attached screen shot.
[image]

datacute, 2020-07-14:
Thanks again @gitterman 
I forgot to update the version in the library.properties file (again)!