datacute/Tiny4kOLED #27: Changing a function for wide range contrast

Deoptim, 2020-07-21:
Hello.
There is a hack in which you can reduce the contrast of the display to the smallest values:

[image]

Video Preview:
https://drive.google.com/file/d/1vcizbV8zYmK-5RECyfEOGb9WmdbUL0Ce/view?usp=sharing

datacute, 2020-08-01:
Hi Deoptim.
I think you intended some additional parenthesis: (contrast + 45) * 1.074 ?
I've found that different screens can have very different responses to the contrast settings.
Using the internal current reference (command AD documented in SSD1306B_v1.1 datasheet) helps to get different devices behaving in a more consistent manner, and the two IREF options result in a big difference in brightness too.
My desire with this library is to keep it as small as possible, so I won't be changing setContrast as that is intended to map straight to the setting of the contrast control register, but I'll think about adding some additional examples.
Also - I suspect user applications are probably better served with a small number of contrast/brightness options, including some really dim settings for adjusting how the display appears in the dark, and some really bright to use outside, and a small number in-between. These are possibly better implemented as small lookup tables for the various settings that affect the brightness.

Deoptim, 2020-08-01:
Hello.
Yes, you may be right.
But I made these settings based on the video from YouTube example:
https://www.youtube.com/watch?v=hFpXfSnDNSY

And as you can see, those settings fit for my display as well.
(I shared the video link above)
Good luck.

SpenceKonde, 2023-02-27:
> I suspect user applications are probably better served with a small number of contrast/brightness options, including some really dim settings for adjusting how the display appears in the dark, and [...]

I agree 100%. It looks like this may already be implemented, at least in some form?