datacute/Tiny4kOLED #29: Scroll display

rtlprmft, 2020-08-28:
Would it be possible to scroll the display one line up when the text reaches extends beyond the bottom of the screen so that the display can act as a mini console? I have looked into the code but I haven't found an easy way... but maybe there is a solution?

rtlprmft, 2020-08-28:
Eventually found the solution. I suggest the following change, either optional or as preprocessor choice:
~~~
void SSD1306Device::newLine(uint8_t fontHeight) {
	oledY+=fontHeight;
        if (oledY > oledPages - fontHeight) {

            const uint8_t off = (oledOffsetY<<3)+8;
            setOffset(0, off);
            setDisplayStartLine(off);

            oledY = oledPages - fontHeight;

            setCursor(0, oledY);
	    clearToEOL();
	}
	setCursor(0, oledY);
}
~~~

datacute, 2020-08-29:
Thanks for asking.
Yes using setDisplayStartLine is the easiest solution, so long as you are not using double-buffering on 128x32 resolution screens.
However in order to keep the library a minimal size, I'd like to keep the methods that most users include to be as small as possible.
If it doesn't change the default behaviour or increase the current code size much, I would consider a patch that allowed different processor directive selected newline implementations, including 'no newline support' to allow users who never send newline characters to save a few bytes.
With pre-compiled libraries, in order to not have all implementations included, pre-processor directive choices can be structured so that the code is in the header files and different implementations passed to the library. I'm interested in other solutions though.

datacute, 2020-08-29:
Also note bug #10 clearToEOL only clears a single page, so the solution you have shown only works for fonts that are a single page high.

rtlprmft, 2020-08-29:
Preprocessor directive is fine. I understand that for some people space is more important than convenience. If this feature is not available you might need an additional buffering which wastes even more space. That is why I believe the general availability would be helpful.  Maybe a callback could do the trick. If you set the callback, it is compiled in. If you do not use it, it is left out. Is that correct? So switching the feature on/off would be with oled.enableTextScroll(oled.callback) and oled.enableTextScroll(nullptr). Of course it needs still space to store the pointer and an if-clause to call to the function. I am not sure how much code that is on a tiny. Certainly the setCursor/clearToEOL could be replaced by a loop.

bctstkla, 2022-08-14:
Thank you @rtlprmft for sharing your solution! This was exactly what I needed, and was a little surprised it wasn't part of the library.

And even more so, thank you @datacute for the awesome library. :)