datacute/Tiny4kOLED #30: Please add support for special Characters

Flo082002, 2020-11-30:
Hi,
Your library is awesome. 
It would be very nice if you could add characters like the German Umlaute  äÄöÖüÜ or the characters with accents or the ° symbol.
Regards 
Florian S

datacute, 2020-12-01:
The library already supports 8-bit fonts, for example the "DatacuteBoxyFont" example includes a custom font with a portion of the codepage 437 characters that are line drawing characters.
You can create a font that includes the rest of codepage 437 to get accented characters. 
I haven't found a nice source to convert.
I might have a go, maybe converting this version, but they are 8x16 pixels: https://retrolcd.com/Curriculum/BitmapFonts

I don't plan on adding Unicode support.

jeroen64, 2021-01-30:
Hi datacute,
Hi Flo082002
Tiny4KOLED works.  I couldn't have made this without it.
[image]

For the build of a stopwatch on a bicycle-steer, I needed a bigger font , like the 16x32. But this one was to big to put 10 characters in a row. the 16x8 was to small. I looked at lots of sites about fonts. The u8g2 fonts turned out not compatible.
The only option left was changing an existing font to 12x24 . I only needed the 10 numbers and a colon.
The comments in the 16x32 font gave me a clue of how to do it.
A character is build in blocks of 8 bits (pixels) high  (3 blocks for a 24pixel high font) and a a certain number of bits with .(12 in my case). spacing is to be added when printing. The project is tested on a Trinket 3v . The aim is to make it a coin-cell project.

[font16x32digits.txt](https://github.com/datacute/Tiny4kOLED/files/5898549/font16x32digits.txt)
This was a .h but somehow github won't let me copy .h files
The app "Fontedit"  from https://kapusta.cc  can help a little ,but not all.

[image]
In this image I show how I digitized my own numbers . This is the "6". The upper right is the first pixel. so the first bits are 00000000 (all dark,but white in the picture) . In hexadecimal this is 00
Converting digital to hexadecimal was done by https://www.rapidtables.com/convert/number/binary-to-hex.html?x=11110000 

only one thing . . .
I could not get the new version renamed.
#include "font16x32digits.h"
const DCfont *currentFont = FONT16X32DIGITS;   // why uppercase  ??

oled.setFont(FONT16X32DIGITS);

I hope this helps creating the glyphs you want.
Regards,        Jeroen

datacute, 2021-01-31:
Thanks Jeroen,

I have another repository with fonts for TIny4kOLED, at https://github.com/datacute/TinyOLED-Fonts

That font repository includes a 16x32 digits font, with a couple more characters than the one you created. (Also defined as FONT16X32DIGITS)

I presume you don't mind me adding your font to that repository? I'll correct the name to be 12x24 instead of 16x32, and will also correct some of the comments that still say that characters are 16 pixels wide.
What's the license? (One main purpose of using a separate repository is that I can keep this one MIT licensed) Did you create the characters yourself, trying to keep the look of the 16x32 ones?

The only reason for the all uppercase is that the Tinusaur library I started with had the fonts named in that way, and I kept backwards compatibility of being able to write  `oled.setFont(FONT8X16);` instead of requiring people to change that to  `oled.setFont(&TinyOLED4kfont8x16);`

Since including a whole lot of fonts in August, I've been thinking about extending my DCFont structure to include an additional piece of information that says how many blank columns are needed between the characters. I will still use fixed width characters, but a lot of fonts use more ram than needed, with each character starting or ending with a zero column for correct kerning.
Most of the fonts added in August do not have that space, as they were designed for use where kerning was added, rather than included in the font data.

An alternative, that I think I'll start with, is to add an additional parameter to the setFont call to specify how many empty columns to write after each character (defaulting to zero).

jeroen64, 2021-01-31:
Hi datacute,

I don't mind you adding my  11 characters  to your repository . I think it is where they belong. 
The "fontedit" app. comes with "Consolas 22p " .Those are the fonts I started with. In this app I edited the glyphs to size and shape,  made a "print-screen" of the raster, and enlarged it in Paint.  All I cared about was the readability from a distance, not the way they look . You may name them any name with 12x24 in it. 
I will take a frequenter look for this repository now.
I am curious  for the way you made all those fonts, though I have no time to make more.

Groet van Jeroen

S-Snowball, 2021-03-19:
I dont understand this also...

How can I print **ü** on the display?

datacute, 2022-01-23:
@Flo082002 and @S-Snowball I've created a runnable example here: https://wokwi.com/arduino/projects/321550584653021779
@jeroen64 Consolas is a font whose license restricts derived works, format conversions, etc.
I've been working on a font converter application (for windows) so that fonts can be converted for use by Tiny4kOLED code.
The demo I link to, has included a font produced by that convertor. The font will be added to https://github.com/datacute/TinyOLED-Fonts soon, but till then, you can copy the contents of the files from that demo.

hb020, 2022-05-31:
Shameless plug here: If you want to create fonts, look at TTF2BMH. My fork (under PR#3) is an improved version that directly supports the format required by Tiny4kOLED (fixed or variable width). It will allow you to take (almost) any TTF font and convert it.

datacute, 2022-06-03:
Here's some screenshots of the font converter application that I've been working on:
[image]
[image]
[image]

hb020, 2022-06-04:
nice work.

datacute, 2022-06-18:
My Font Converter windows application is now available at https://github.com/datacute/FontConverter