datacute/Tiny4kOLED #31: Wont compile on Arduino IDE 1.8.13.

facelessloser, 2020-12-26:
So I installed the library thought the IDE but when I got to compile it gives me these errors. Ive installed the Attiny85 boards files from here https://create.arduino.cc/projecthub/arjun/programming-attiny85-with-arduino-uno-afb829

Arduino: 1.8.13 (Linux), Board: "ATtiny25/45/85, ATtiny85, Internal 16 MHz"

/home/facelessloser/arduino-1.8.13/hardware/arduino/avr/libraries/Wire/src/utility/twi.c: In function 'twi_init':
/home/facelessloser/arduino-1.8.13/hardware/arduino/avr/libraries/Wire/src/utility/twi.c:88:16: error: 'SDA' undeclared (first use in this function)
   digitalWrite(SDA, 1);
                ^~~
/home/facelessloser/arduino-1.8.13/hardware/arduino/avr/libraries/Wire/src/utility/twi.c:88:16: note: each undeclared identifier is reported only once for each function it appears in
/home/facelessloser/arduino-1.8.13/hardware/arduino/avr/libraries/Wire/src/utility/twi.c:89:16: error: 'SCL' undeclared (first use in this function); did you mean 'SPL'?
   digitalWrite(SCL, 1);
                ^~~
                SPL
In file included from /home/facelessloser/arduino-1.8.13/hardware/tools/avr/avr/include/avr/io.h:99:0,
                 from /home/facelessloser/arduino-1.8.13/hardware/arduino/avr/libraries/Wire/src/utility/twi.c:26:
/home/facelessloser/arduino-1.8.13/hardware/arduino/avr/libraries/Wire/src/utility/twi.c:92:7: error: 'TWSR' undeclared (first use in this function); did you mean 'ACSR'?
   cbi(TWSR, TWPS0);
       ^
/home/facelessloser/arduino-1.8.13/hardware/arduino/avr/libraries/Wire/src/utility/twi.c:92:3: note: in expansion of macro 'cbi'
   cbi(TWSR, TWPS0);
   ^~~
/home/facelessloser/arduino-1.8.13/hardware/arduino/avr/libraries/Wire/src/utility/twi.c:92:13: error: 'TWPS0' undeclared (first use in this function); did you mean 'TWSR'?
   cbi(TWSR, TWPS0);
             ^
/home/facelessloser/arduino-1.8.13/hardware/arduino/avr/libraries/Wire/src/utility/twi.c:92:3: note: in expansion of macro 'cbi'
   cbi(TWSR, TWPS0);
   ^~~
/home/facelessloser/arduino-1.8.13/hardware/arduino/avr/libraries/Wire/src/utility/twi.c:93:13: error: 'TWPS1' undeclared (first use in this function); did you mean 'TWPS0'?
   cbi(TWSR, TWPS1);
             ^
/home/facelessloser/arduino-1.8.13/hardware/arduino/avr/libraries/Wire/src/utility/twi.c:93:3: note: in expansion of macro 'cbi'
   cbi(TWSR, TWPS1);
   ^~~
/home/facelessloser/arduino-1.8.13/hardware/arduino/avr/libraries/Wire/src/utility/twi.c:94:3: error: 'TWBR' undeclared (first use in this function); did you mean 'TWSR'?
   TWBR = ((F_CPU / TWI_FREQ) - 16) / 2;
   ^~~~
   TWSR
/home/facelessloser/arduino-1.8.13/hardware/arduino/avr/libraries/Wire/src/utility/twi.c:102:3: error: 'TWCR' undeclared (first use in this function); did you mean 'TWBR'?
   TWCR = _BV(TWEN) | _BV(TWIE) | _BV(TWEA);
   ^~~~
   TWBR
In file included from /home/facelessloser/arduino-1.8.13/hardware/tools/avr/avr/include/avr/io.h:99:0,
                 from /home/facelessloser/arduino-1.8.13/hardware/arduino/avr/libraries/Wire/src/utility/twi.c:26:
/home/facelessloser/arduino-1.8.13/hardware/arduino/avr/libraries/Wire/src/utility/twi.c:102:14: error: 'TWEN' undeclared (first use in this function); did you mean 'TWCR'?
   TWCR = _BV(TWEN) | _BV(TWIE) | _BV(TWEA);
              ^
/home/facelessloser/arduino-1.8.13/hardware/arduino/avr/libraries/Wire/src/utility/twi.c:102:26: error: 'TWIE' undeclared (first use in this function); did you mean 'TWEN'?
   TWCR = _BV(TWEN) | _BV(TWIE) | _BV(TWEA);
                          ^
/home/facelessloser/arduino-1.8.13/hardware/arduino/avr/libraries/Wire/src/utility/twi.c:102:38: error: 'TWEA' undeclared (first use in this function); did you mean 'TWEN'?
   TWCR = _BV(TWEN) | _BV(TWIE) | _BV(TWEA);
                                      ^
/home/facelessloser/arduino-1.8.13/hardware/arduino/avr/libraries/Wire/src/utility/twi.c: In function 'twi_disable':
/home/facelessloser/arduino-1.8.13/hardware/arduino/avr/libraries/Wire/src/utility/twi.c:114:3: error: 'TWCR' undeclared (first use in this function); did you mean 'EECR'?
   TWCR &= ~(_BV(TWEN) | _BV(TWIE) | _BV(TWEA));
   ^~~~
   EECR
In file included from /home/facelessloser/arduino-1.8.13/hardware/tools/avr/avr/include/avr/io.h:99:0,
                 from /home/facelessloser/arduino-1.8.13/hardware/arduino/avr/libraries/Wire/src/utility/twi.c:26:
/home/facelessloser/arduino-1.8.13/hardware/arduino/avr/libraries/Wire/src/utility/twi.c:114:17: error: 'TWEN' undeclared (first use in this function); did you mean 'TWCR'?
   TWCR &= ~(_BV(TWEN) | _BV(TWIE) | _BV(TWEA));
                 ^
/home/facelessloser/arduino-1.8.13/hardware/arduino/avr/libraries/Wire/src/utility/twi.c:114:29: error: 'TWIE' undeclared (first use in this function); did you mean 'TWEN'?
   TWCR &= ~(_BV(TWEN) | _BV(TWIE) | _BV(TWEA));
                             ^
/home/facelessloser/arduino-1.8.13/hardware/arduino/avr/libraries/Wire/src/utility/twi.c:114:41: error: 'TWEA' undeclared (first use in this function); did you mean 'TWEN'?
   TWCR &= ~(_BV(TWEN) | _BV(TWIE) | _BV(TWEA));
                                         ^
/home/facelessloser/arduino-1.8.13/hardware/arduino/avr/libraries/Wire/src/utility/twi.c:117:16: error: 'SDA' undeclared (first use in this function)
   digitalWrite(SDA, 0);
[... 165 lines cut ...]
/home/facelessloser/arduino-1.8.13/hardware/arduino/avr/libraries/Wire/src/utility/twi.c:510:7: error: 'TWDR' undeclared (first use in this function); did you mean 'TWSR'?
       TWDR = twi_slarw;
       ^~~~
       TWSR
/home/facelessloser/arduino-1.8.13/hardware/arduino/avr/libraries/Wire/src/utility/twi.c:530:10: error: 'TWCR' undeclared (first use in this function); did you mean 'TWDR'?
          TWCR = _BV(TWINT) | _BV(TWSTA)| _BV(TWEN) ;
          ^~~~
          TWDR
In file included from /home/facelessloser/arduino-1.8.13/hardware/tools/avr/avr/include/avr/io.h:99:0,
                 from /home/facelessloser/arduino-1.8.13/hardware/arduino/avr/libraries/Wire/src/utility/twi.c:26:
/home/facelessloser/arduino-1.8.13/hardware/arduino/avr/libraries/Wire/src/utility/twi.c:530:21: error: 'TWINT' undeclared (first use in this function)
          TWCR = _BV(TWINT) | _BV(TWSTA)| _BV(TWEN) ;
                     ^
/home/facelessloser/arduino-1.8.13/hardware/arduino/avr/libraries/Wire/src/utility/twi.c:530:34: error: 'TWSTA' undeclared (first use in this function); did you mean 'TWS3'?
          TWCR = _BV(TWINT) | _BV(TWSTA)| _BV(TWEN) ;
                                  ^
/home/facelessloser/arduino-1.8.13/hardware/arduino/avr/libraries/Wire/src/utility/twi.c:530:46: error: 'TWEN' undeclared (first use in this function); did you mean 'TWINT'?
          TWCR = _BV(TWINT) | _BV(TWSTA)| _BV(TWEN) ;
                                              ^
exit status 1
Error compiling for board ATtiny25/45/85.

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

cybergenesis70, 2021-03-01:
Any solution to this?  I am having the same issue. Thanks

fabiougolini, 2021-03-01:
Using ATTiny board from this library
http://drazzy.com/package_drazzy.com_index.json
will compile correctly. But I am not able to make my display work with the ATTiny85 anyway.. :(

facelessloser, 2021-03-01:
I haven't heard anything back on this issue. I would love to get it working but I don't know where to start

datacute, 2021-03-05:
The errors look like they might be because the package you are using to support the ATTiny series doesn't include support for the Wire library (for I2C communication), and arduino's wire library for avr chips is being used.
The board manager you are using hasn't been updated in 6 years.
I use the ATTinyCore from Spence Konde, which includes an implementation of Wire.
The link to the drazzy package is what you add to your "Additional Board Manager Urls" in Arduino IDE preferences to get access to Spence Konde's ATTiny support.

Alternatively, The Tiny4kOLED library will select Adafruit's TinyWireM or Technoblogy's tiny-i2c instead of the Wire library, if they are included before Tiny4kOLED.h.
I don't know whether they will work with your board manager.

SpenceKonde, 2021-03-22:
It is worth noting that under the hood, what is the Wire.h library for ATTinyCore? Well... actualllly.... It's a bigass bunch of #ifdefs.... and copies of ~three~ four other libraries, modified so they give the class the same name as Wire, and with the methods tweaked as necessary to have the same signatures as Wire.h's methods....I used, i think, USIWire, not TinyWireM as the base for USI I2C mode on parts with a USI (I think because I wanted it to be truly transparent - meaning slave too, not master only) , stock Wire for the ATtiny88 and... I don't think anything else, and a full on software I2C master for the two parts that only have slave TWI in hardware (the x41 and 828)... with a fourth library, WireS pulled in to get slave on them (obviously I had to mash WireS and SoftI2CMaster or whatever software I2C implementation I used together into one class.. I couldn't ever get the compiler to not compile slave functions if slave wasn't used and vise versa, so I wound up just copping out and putting a tools submenu to select if you want slave, master or both. 

The library works on megaTinyCore-supported 0/1-series (and presumably 2-series) parts (with Wire.h, albeit an entirely different Wire.h, which is basically cribbed from nano every core). I do believe it works anywhere that has a wire library that does what it's f'ing supposed to. I would suggest that, unless your testing refutes this theory, that you clarify that it's not just for classc Tiny parts. I mean in a few months, there are going to be tinyAVR 2-series parts with more RAM than a '328p - and just as much flash (though flash doesn't go quiiiite as far; not sure how large the penalty is under typical code composition, but the register layout is less favorable on the modern AVRs)

SpenceKonde, 2023-02-27:
I think this issue can probably be closed because as you clearly state in the docs, Wire.h as the method of data transfer is supported for the ATtiny85 (as well as all other devices with a USI instead if a TWI), *unless* using my ATTinyCore, which provides a compatible Wire.h implementation for the USI parts as well. Several other options are presented which will work when using other cores.

closed 2023-02-28 by datacute