datacute/Tiny4kOLED #32: Add SH1106 132x64 display support

Deoptim, 2020-12-27:
Hello.
I post this adapted library (ad58375 on Aug 29) for the SH1106 controller.
Made it possible not to use (delete) unsupported functions if the parameter "#define SH1106" is defined in the Tiny4kOLED_common.h file.
Applicable for SH1106 controller as there are slight differences from SSD1306.
I made this library for myself.
Thanks.
[Tiny4kOLED-ad58375.zip](https://github.com/datacute/Tiny4kOLED/files/5744255/Tiny4kOLED-ad58375.zip)

hamedmalekii, 2022-05-09:
> Hello. I post this adapted library ([ad58375](https://github.com/datacute/Tiny4kOLED/commit/ad583754b8521c922e7def1a85a2866a8790e6d3) on Aug 29) for the SH1106 controller. Made it possible not to [...]

how can use your castum library?
i use inport your library and used sh1106 display not chenged?
The corner of the combo image still has pixel clutter
[image]

hamedmalekii, 2022-05-09:
[image]

Deoptim, 2022-05-10:
@hamedmalekii Hello,
Are you use this initial sequence:
```
oled.begin(132, 64, sizeof(tiny4koled_init_132x64), tiny4koled_init_132x64);
oled.clear();
oled.on();
oled.init();
```
?

Note that the width here is 132 instead of 128...

hamedmalekii, 2022-06-20:
@Deoptim 
i use this order (132, 64, sizeof(tiny4koled_init_132x64), tiny4koled_init_132x64); but the display is upside down
Is there a way to solve it?

hamedmalekii, 2022-06-21:
> @hamedmalekii Hello, Are you use this initial sequence:
> 
> ```
> oled.begin(132, 64, sizeof(tiny4koled_init_132x64), tiny4koled_init_132x64);
> oled.clear();
> oled.on();
> oled.init();
> ```
> 
> ?
> 
> Note that the width here is 132 instead of 128...
@datacute hi
i use this order (132, 64, sizeof(tiny4koled_init_132x64), tiny4koled_init_132x64); but the display is upside down
Is there a way to solve it?

Deoptim, 2022-06-21:
@hamedmalekii
The first thing I advise you to do is to check the display that it works at all - on Arduino with some library and take an example of rendering.
[image]

datacute, 2022-06-22:
@hamedmalekii are you using Deoptim's code or the latest version of my code?
The latest version of my code will probably mostly work if you use the begin method that specifies the offsets and size, but use the init sequence for the 128x64 SSD1306:
```
oled.begin(2, 0, 132, 64, sizeof(tiny4koled_init_128x64), tiny4koled_init_128x64);
```
There is also a function to rotate 180 degrees:
```
oled.setRotation(1)
```
(try both 1 and 0)

hamedmalekii, 2022-06-25:
> @hamedmalekii are you using Deoptim's code or the latest version of my code? The latest version of my code will probably mostly work if you use the begin method that specifies the offsets and size, [...]

thanks
i try it

hamedmalekii, 2022-07-30:
> @hamedmalekii are you using Deoptim's code or the latest version of my code? The latest version of my code will probably mostly work if you use the begin method that specifies the offsets and size, [...]

dont work
after use  oled.setRotation(1)
i have this error

**exit status 1
expected ';' before 'oled'**

i used indentify example sketch and  to slide 4 evrything is ok
What are the display settings related to slide 4?

`
/*
 * Tiny4kOLED - Drivers for SSD1306 controlled dot matrix OLED/PLED 128x32 displays
 *
 * Based on ssd1306xled, re-written and extended by Stephen Denne
 * from 2017-04-25 at https://github.com/datacute/Tiny4kOLED
 *
 * This example attempts to identify your screen.
 * When the screen shows a rectangle with a number in it, 
 * find that number in this source file to see how to use the
 * Tiny4kOLED library with your screen.
 *
 */

#include <Tiny4kOLED.h>

#define pinPrevious 3
#define pinNext 4

// The values shown here are the default values at reset.
// As this example changes lots of values,
// this sequence is sent between each init sequence
// The screen SHOWN as sequence 0 is sending this sequence,
// then the libraries default sequence (for a 128 x 32 screen)
static const uint8_t ssd1306_init_sequence0 [] PROGMEM = {  // Initialization Sequence
  0xAE,         // Display OFF (sleep mode)
  0x2E,         // Disable scroll
  0x20, 0b10,   // Set Memory Addressing Mode
                // 00=Horizontal Addressing Mode; 01=Vertical Addressing Mode;
                // 10=Page Addressing Mode (RESET); 11=Invalid
  0xB0,         // Set Page Start Address for Page Addressing Mode, 0-7
  0xC0,         // Set COM Output Scan Direction
  0x00,         // Set low nibble of column address
  0x10,         // Set high nibble of column address
  0x40,         // Set display start line address
  0x81, 0x7F,   // Set contrast control register
  0xA0,         // Set Segment Re-map. A0=column 0 mapped to SEG0; A1=column 127 mapped to SEG0.
  0xA6,         // Set display mode. A6=Normal; A7=Inverse
  0xA8, 0x3F,   // Set multiplex ratio(1 to 64)
  0xA4,         // Output RAM to Display
                // 0xA4=Output follows RAM content; 0xA5,Output ignores RAM content
  0xD3, 0x00,   // Set display offset. 00 = no offset
  0xD5, 0x80,   // --set display clock divide ratio/oscillator frequency
  0xD9, 0x22,   // Set pre-charge period
  0xDA, 0x12,   // Set com pins hardware configuration
  0xDB, 0x20,   // --set vcomh 0x20 = 0.77xVcc
  0xAD, 0x00,   // Select external IREF
  0x8D, 0x10    // DC-DC disabled
};

// Initialization sequence for 128 x 32 screen
static const uint8_t ssd1306_init_sequence1 [] PROGMEM = {
  0xA8, 0x1F,   // Multiplex ratio
  0xDA, 0x02,   // HW pin configuration
  // If you are copying this, include the common_post_init_sequence values to here
};

// Initialization sequence for rotated 128 x 32 screen
[... 164 lines cut ...]
  uint8_t endSCrollPage = startScrollPage + 1;
  uint8_t startScrollColumn = 8 + minx;
  uint8_t endScrollColumn = startScrollColumn + width - 16;
  for (uint8_t x = 0; x < width - 16; x++)
  {
    delay(50);
    if (digitalRead(pinPrevious) == 0) {
      if (screen == 0) screen = numScreens - 2;
      else screen -= 2;
      break;
    }
    if (digitalRead(pinNext) == 0) break;
 
  }
}

void nextScreen() {
  screen += 1;
  if (screen == numScreens) screen = 0;
  minx = screens[screen].minx;
  miny = screens[screen].miny;
  width = screens[screen].width;
  height = screens[screen].height;
}
`

SpenceKonde, 2023-02-27:
This issue clearly appears to be a missing `;` on the line immediately preceding that line. It has likely been solved by now, and is probably closable.

miamijerry, 2025-04-18:
I tried Deoptim's ad58375 for a 1.3" SH1106 OLED display, which worked fine. 

Amazingly I tried a 2.42" SSD1309 OLED display and it also worked.

Fantastic,
Thank you