datacute/Tiny4kOLED #33: Arduino simulation pages for Tiny4kOLED library?

Puneethkumarsn, 2021-03-16:
Hello Datacube

good day. We have created a permanent link for the Tiny4kOLED simulation.
Please have a look here: 
https://wokwi.com/arduino/projects/293148150899671560

I would be glad to create a PR for the readMe file to include the link. Not only for this particular example but also for the other examples. This will be a great chance for your readers to see your library in action 😃

Thank you
Puneeth

datacute, 2021-03-20:
That's incredible work! I'd definitely be interested in a PR with links.

Puneethkumarsn, 2021-03-20:
Thanks a lot. I will be back soon. Thanks for the motivating words :) All credits go to the wokwi Arduino simulator.

datacute, 2021-04-05:
Merged

closed 2021-04-05 by datacute