datacute/Tiny4kOLED #35: how to draw cyrillic?

lancer-1, 2021-04-07:


datacute, 2021-04-19:
Tiny4kOLED only supports single byte 8 bit encodings, not unicode.

Do you want KOI8-R or windows 1251 or codepage 866?

Here's a useful page for converting strings:
http://string-functions.com/encodedecode.aspx

datacute, 2021-04-23:
I'm also in the middle of adding support for proportional width fonts, and thinking that I might be able to easily add support for multiple unicode blocks. While I'm familiar with unicode, I've not worked with it in c nor arduino before. Do you know whether utf-8 or utf-16 is more common for cyrillic?

lancer-1, 2021-04-23:
Yes, of course, it's will be useful!

lancer-1, 2021-04-26:
> Do you know whether utf-8 or utf-16 is more common for cyrillic?

Windows 1251, but I think there are even better.

closed 2021-04-26 by lancer-1

datacute, 2021-05-16:
Here's an image of my first working version.

[image]

datacute, 2021-05-17:
Some background...

The image in my previous comment was from code I've written that understands utf-8.
The user API (at the moment) is just one new font setting method.
```
oled.setUnicodeFont(FONTSEGOEUIREGULAR13);
oled.println(F("Кириллические"));
oled.println(F("шрифты."));
oled.println(F("«Приветствую мир»"));
oled.println(F("    16 мая 2021 г."));
```
The unicode font references four smaller font portions, each storing glyphs for a range of characters in a unicode plane and block, U+0021 to U+003F, U+00A1 to U+00BF, U+0400 to U+045F, and U+2010 to U+2027.
It's showing the Segoe UI font, so not distributable. I'll have to build up my font conversion code into a UI to allow conversion of fonts that people have the rights to convert and use for their own purposes. My conversion application is windows specific, and not yet available.