datacute/Tiny4kOLED #36: Vertical Lines on cheap SSD1306 0.91" OLED

sl6der, 2021-08-29:
Good day,

i´m struggeling on a problem with some chinese 0.91OLEDs and the Tiny4kOLED Library.
Overall at first i tested the U8G2 Lib to verify the general health of my oled and got a positiv result. Every graphics test was successful.
After switching from the Arduino Pro Mini to the Attiny85 I have to shrink the code and found this Tiny4kOLED Library. So now i´m stuck on a working code with some issues on the OLED.

[image]

At first i tested the health with the example codes and already there it seems something was happening with the output on the Oled.

I tried to fix it with Tiny4kOLED_Wire.h instead of Tiny4kOLED.h, but the result was the same.

Did i missed something? 
The TinyWireM Library is at the latest version 1.1.0.

Can anybody help me?

datacute, 2021-08-30:
One way Tiny4kOLED save's space is by only sending initialization parameters that are different from the default.

It might be that the screen has kept some settings from the other library.

To check this, there are alternative oled.begin calls, one of which supplies all the documented defaults.

Once the screen is back to is default settings, a smaller begin call can be used again.

sl6der, 2021-08-30:
It seems I was able to fix it.

Instead of Tiny4kOLED_Wire.h I used Tiny4kOLED_TinyWireM.h and for Init i used 
oled.begin(128, 32, sizeof(tiny4koled_init_defaults), tiny4koled_init_defaults);
oled.enableChargePump();

Now all lines are gone and I can use the simple oled.begin(); Method to initialize the display.

Thanks for your help datacute ;-)

closed 2021-08-30 by sl6der