datacute/Tiny4kOLED #37: Make printDoubleSize more convenient

fijam, 2021-09-05:
I'm a total noob when it comes to c/c++ so bear with me please, but from what I understand `oled.printDoubleSize()` and `oled.printDoubleSizeSmooth()` either takes a string from flash with the F macro or needs to be fed byte-by-byte. Would it be possible to make it behave like regular `oled.print()` so I could easily print a variable? Or perhaps add an example demonstrating the right way to do it?

datacute, 2021-09-05:
You are right - there isn't at the moment a way to use it other than directly calling those methods.

datacute, 2021-09-25:
I started a wiki page to record some details about this.
https://github.com/datacute/Tiny4kOLED/wiki/Refactoring-to-support-double-sized-rendering

datacute, 2021-12-31:
I just released v2.2.0 where the `printDoubleSize` and `printDoubleSizeSmooth` methods are replaced with `setFontX2`, `setFontX2Smooth`, `setUnicodeFontX2`, and `setUnicodeFontX2Smooth` so that the full variety of normal print methods can work with double sized text.

closed 2021-12-31 by datacute