datacute/Tiny4kOLED #41: switchFrame vs setUnicodeFontX2?

outdever, 2022-01-25:
Hello. There is a need to use both of these functions together. When applied at the same time, the first area is not drawn. Example of non-working code:
```
#define TINY4KOLED_NO_PRINT
#define SDA_PIN 6
#define SCL_PIN 4

#include <Tiny4kOLED_bitbang.h>
//#include <USIWire.h>
//#include <Tiny4kOLED.h>

uint8_t width = 128;
uint8_t height = 32;

// Fonts from this TinyOLED-Fonts library

// 2386 bytes in font + 3184 bytes for the program = 5570 byte total
#include "Open_Sans_Light_13_5_Unicode.h"
const DCUnicodeFont *currentFont = FONTOPENSANSLIGHT135;
//#define SAMPLE "Кириллические\nшрифты.\n"
#define SAMPLE "ЗАРЯД\n"

uint8_t currentCharacter;
uint8_t prc=0;

void setup() {
  // put your setup code here, to run once:
  oled.begin(width, height, sizeof(tiny4koled_init_128x32br), tiny4koled_init_128x32br);
 
  //oled.begin(128, 64, sizeof(tiny4koled_init_defaults), tiny4koled_init_defaults);
  oled.enableChargePump();
  oled.setRotation(1);
  oled.setUnicodeFont(currentFont);
  //oled.setUnicodeFontX2(currentFont);
  
  oled.on();
  oled.switchRenderFrame();
}

void loop() {
  drawScreen();
  delay(500);
  oled.switchFrame();
  //while(1);
}

void drawScreen() {
  //oled.setCursor(0, 0);
  oled.clear();
  oled.setUnicodeFontX2Smooth(currentFont);
  oled.setCursor(0, 0);
  
#ifndef TINY4KOLED_NO_PRINT
  oled.println(F(SAMPLE));

#else
  PGM_P p = reinterpret_cast<PGM_P>(F(SAMPLE));
  while (1) {
    byte c = pgm_read_byte(p++);
    if (c == 0) break;
    oled.write(c);
  }
#endif
}
```

datacute, 2022-01-25:
Thanks for reporting that. I think I know the reason, and will work on a fix.

datacute, 2022-01-26:
@outdever can you please test with version 2.2.2?

outdever, 2022-01-26:
Thank you very much! Everything works great

closed 2022-01-26 by outdever