datacute/Tiny4kOLED #42: How to tune the cursor for y ONE pixle by ONE pixle? x is ok but y is 1 PAGE by 1 PAGE.

sonictl, 2022-03-14:
The function  `oled.setOffset(1,5); ` will offset the x,y coordinate to x+1, y+5*8
What if I want offset of x+1, y+5?
The end coordinate of y (y1 in program) is setting by adding 8 pixles each 1 number. This is an inhuman design. Isn't it?

datacute, 2022-03-14:
The SSD1306 uses 'pages' of memory. You cannot write a single pixel, but are always setting 8 pixels at once.
The setOffset function was added with setWidth and setHeight, to define the area of RAM that the hardware displays.
Since hardware will be mapping whole pages of RAM onto its pixels, the height and y parameter of setOffset are expected to be multiples of 8. The library divides by 8 to get which page number to use.

sonictl, 2022-03-20:
With the experience of using Adafruit_SSD1306, it's smooth to set the start point of y as ONE pixcel by ONE pixcel, when drawing bitmap. That's convienient.
I suggest to wrap this lib by a layer above, in order to freely set the start point of y coord for bitmap. Just like Adafruit_SSD1306.
Tiny4kOLED occupies less flash memory, this is COOL for ATtiny chip users.
On ATMega32U4 chip, with larger flash memory, it's possible to use Adafruit_SSD1306.

sonictl, 2022-03-20:
by the way, the horizontal bitmap data generation is Better than the vertical, which is adoptted by this lib. The vertical mechanism meets the convenience of chip but not the human and tools such as [drawing2bitmap webtool](http://dot2pic.com/). Even though I did my bitMap designing for this lib with [image2cpp](https://diyusthad.com/image2cpp), it's more work for preparing the png for that `image2cpp` webtool. Sometimes, direct drawing is preferred.

datacute, 2022-04-22:
This library is extremely optimised for space, so doesn't have an in-RAM buffer that is drawn to, which is then completely copied to  the screen's memory. The low level abilities are exposed, to start sending data, write a whole lot of data, then stop sending data, so you can write your own bitmap drawing functions, and do not need to use the bitmap drawing routine that the library provides. I'm happy to consider pull requests if you come up with a good solution.

closed 2022-04-22 by datacute