datacute/Tiny4kOLED #44: Does not work with SSD1316.

7-rate, 2022-05-22:
I have tried using the SSD1316 with an ATtiny1604, but it does not work properly.
Specifically, the display turns into a sandstorm.

Is there any way to make it work?

This is the OLED I used.
https://www.alibaba.com/product-detail/0-87-Micro-Screen-Mini-Thin_1600342776585.html?spm=a2756.trade-order-form-buy.0.0.3c28mdSqmdSqFZ

datacute, 2022-06-03:
I took a quick look at the docs of the 1316 compared to the docs on the 1306, they look like pretty much the same commands, so portions of this library might be able to work with it. Try experimenting with different initialization sequences. The difference in the number of rows means maybe the MUX setting should be different.

7-rate, 2022-06-28:
It worked fine.
It seems that the cause of the sandstorm was a hardware problem.
(There were not enough capacitors on the RES pin)

closed 2022-06-28 by 7-rate