datacute/Tiny4kOLED #46: Is there more documentation somewhere?

IdontKnowWhatToNameThisAccount, 2022-07-21:
There are a lot of functions used in the examples that arent properly explained or documented anywhere, or at least I cant seem to find them. For example I dont really understand how oled.fill is supposed to work. Is there some documentation that simply explains the functionality of each function?

datacute, 2022-07-25:
Most functions are simply exposing commands of the SSD1306. These can be found by searching google for:
ssd1306 datasheet "rev 1.5"

SpenceKonde, 2023-02-27:
I would suggest linking to a copy of the latest datasheet referenced while working on development of this library. Unless the manufacturer keeps moving it around; Microchip now does this constantly for the AVRs, which is why I don't link directly to AVR datasheets anymore; I link to the Microchip product pages (mTC and DxC only currently) - where the latest, usually working - links are to be found.)

SpenceKonde, 2023-02-27:
Okay, yeah, this also needs an API reference. Right now now I don't know how anyone does anything with this library? Do perform a deep dive on the source and write their own documentation and not share it? Do they just blindly copy-paste example code cargo-cult style then tweak it? (which of course is well known as a development anti-pattern)? 

Here's what the path ahead of me looks like. 

`(what I want to do) -XXX- (API calls to make)------(data that gets set to screen)-----(what I want done)`

And I've got a lot of these screens... 
[image]