datacute/Tiny4kOLED #48: Non-Blocking when OLED disconnected?

IAmOrion, 2022-11-10:
This may seem like an odd usage case scenario, but I have a small ATTiny85 circuit, where the OLED screen is "optional"
The problem, I just discovered, is that if the OLED is *not* connected, nothing works! I assume it's preventing the sketch from continuing as the OLED was not detected.  

Adafruit do this using this in the setup()
```
  if(!display.begin(SSD1306_SWITCHCAPVCC, SCREEN_ADDRESS)) {
    Serial.println(F("SSD1306 allocation failed"));
    for(;;); // Don't proceed, loop forever
  }
```
But of course, you can comment out or just remove the for(;;); part, and everything works and continues as normal even when the OLED is disconnected.

I can't seem to find where this Tink4KOLED library is doing this similar action? I presume it must be because, as I mentioned, if the OLED is disconnect, then it "freezes" - nothing continues.

So can I go about using this library in a non-blocking fashion? So that if the OLED is not connected, everything carries on as normal!?

datacute, 2022-11-10:
The simple solution, depending on which OLED you have, and which I2C implementation you are using, is to
`#define TINY4KOLED_QUICK_BEGIN`
before including Tiny4kOLED.

If you search for that in this library's source code, you can see how that should work, and you could write a test of whether an OLED is connected.

IAmOrion, 2022-11-10:
Ah, yeah, I did find that, but all my OLEDs are Blue (since that's the most common I guess) so didn't go with that as it states it's for White OLEDs.

I'll try and investigate how to check if an OLED is connected.  I guess looking at the basic I2C Scanner examples would be a good start (Although I'm using TinyI2C rather than Wire)

closed 2022-11-11 by IAmOrion

datacute, 2022-11-11:
I'm pleased you got it working.
With TinyI2C , when `TinyI2c.start` returns true, you need to call TinyI2c.stop.