datacute/Tiny4kOLED #49: Is there a way to mirror the output?

phillmj, 2023-01-21:
I've got a screen that is an SSD1312, and it works with the Adafruit 1306 library. I had to use some commands to mirror the screen with the Adafruit library, and it works fine.

The issue I'm having is whenever I try to use commands from Tiny4kOLED, the output is mirrored again.

Is there a command that will mirror the screen output? All the characters are backwards and left/right like from a mirror; see screen shot.

Love your library and fonts. I would really like to use them in my project. Any thoughts?

Thanks,

Mike
[image]

phillmj, 2023-01-21:
OK, I read some of your other posts, and you said to go into the data sheet for SSD1306.

I did, and I came up with the following, which works great:

```
oled.setMultiplexRatio(0xA8);
oled.setSegmentRemap(0xA1); 
oled.setComOutputDirection(0xC8);
```

closed 2023-01-21 by phillmj

datacute, 2023-01-23:
@phillmj I'm glad you found a solution.
The SSD1306 will remember many settings not explicitly cleared or otherwise set in the initialization routine, and to save as many bytes as possible, I default to just about minimal initialization routines.
Having corrected your display, you might not need those command in your sketch anymore,

phillmj, 2023-01-23:
Hi Stephen, thanks for the additional info, very helpful.

Unfortunately, I need to use commands from both Tiny4kOLED and the Adafruit library. And when I use one, it inverts the other! It's kind of a nightmare; these two are not playing well together at all with the screen I have.

They worked perfectly together with the actual SSD1306 screen I have; the problem is this odd SSD1312 screen that uses SSD1306 commands, but it's mirrored and flipped by default.

In any case, I'll probably have to use a different screen, because these this screen does not play well with these two libraries together.

Thanks again,

Mike

datacute, 2023-01-25:
@phillmj I see.
The library supports supplying your own initialisation sequence.
There are a lot of examples of different init sequences here:
https://github.com/datacute/Tiny4kOLED/blob/master/src/Tiny4kOLED_common.h#L231

An example of using one is here:
https://github.com/datacute/Tiny4kOLED/blob/master/examples/Devices/128x32/128x32.ino#L19
`oled.begin(width, height, sizeof(tiny4koled_init_128x32br), tiny4koled_init_128x32br);`

Just calling `oled.begin()` results in:
https://github.com/datacute/Tiny4kOLED/blob/master/src/Tiny4kOLED.cpp#L144
`begin(sizeof(tiny4koled_init_128x32r), tiny4koled_init_128x32r);`

Many of the init sequences I have provided already include mirroring of the display, along with flipping top to bottom, resulting in a rotation.
You could create and use an initialisation sequence for the SSD1312.

datacute, 2023-01-25:
Adafruit's init sequence is implemented here:
https://github.com/adafruit/Adafruit_SSD1306/blob/master/Adafruit_SSD1306.cpp#L564

This includes
`SSD1306_SEGREMAP | 0x1` which is 0xA1
`SSD1306_COMSCANDEC` which is 0xC8

datacute, 2023-01-25:
One more message for today: your calls shown in an earlier message to setSegmentRemap and setComOutputDirection though not needed if you supply an init sequence... they are not being supplied with the correct parameters. They each take 0 or 1.
setSegmentRemap will still work, as the lowest bit is kept, but setComOutputDirection takes the lowest bit and shifts it, so supplying 0xC8 is the same as supplying 0, and results in sending the command 0xC0. To send 0xC8 you need to supply that method with the parameter value of 1.

phillmj, 2023-01-25:
Thanks so much Stephen, and for the correction too.

I can get the display to work correctly as long as I don't mix together both your library commands beginning with "oled." and the Adafruit library beginning with "display."

As long as I stick to one or the other it's fine. But I wanted to use your fonts with the Adafruit library, and that's where the trouble begins. If I initialize with the Adafruit commands and stick to Adafruit, it's fine. But the moment I start using your fonts the screen mirrors again.

I will experiment further and if I get it working I'll post back.

Thanks for your terrific library and your wonderful support.

Mike