datacute/Tiny4kOLED #55: Please check the % character font.

rom-far, 2023-03-03:
The displayed % characters appear to be mirrored.
Could you please check the font header file?
Only font6x8.h has been confirmed.
thank you.
I'm sorry if I made a mistake.

rom-far, 2023-03-03:
additional information. Only when ``oled.setFontX2(FONT6X8);'' is specified, the % character becomes a mirror character, and when ``oled.setFont(FONT6X8);'' is specified, the % character is displayed normally.

rom-far, 2023-03-03:
It seems that this issue is not related to the font6x8.h file.

datacute, 2023-04-02:
Thanks for reporting this. It is a fault in the FONT6X8 font, and in the proportional version I generated from it: FONT6X8P, it is simply easier to see when double size.
Here's an online simulator showing the problem: https://wokwi.com/projects/360858143862483969
I'll create a fix.

datacute, 2023-04-02:
Other libraries have found this too, e.g
https://github.com/lexus2k/ssd1306/issues/23

datacute, 2023-04-30:
Fixed in v2.3

closed 2023-04-30 by datacute