datacute/Tiny4kOLED #56: Why horizontal setting should seperate to 8

B3B3K, 2023-07-22:
I cant handle the hole pixels for wake cause horizontal miss, how i can change .cpp file to write to directly in y coordinate?
(I had tried it about remove offset but idk what is the mean off offset in this case, lol my fault but i have miss )

datacute, 2023-07-23:
The hardware requires overwriting 8 pixels (1 byte) at a time, and doesn't allow reading what is already set.
To work out the y coordinate, you need the high bits for the 'page' number, and low 3 bits to identify the bit within the byte, however you need to set or clear all 8 bits in the byte to the required values before you send it to the screen.

B3B3K, 2023-08-20:
That means, when i setting coordinate on oled and placing pixel, i should think about its hex value (for serious :D measurements :D), right?