datacute/Tiny4kOLED #57: Flicking - I'm so tierd to fix it, help me please

Xnork, 2023-09-23:
Look at 
```c++
/*
 * Tiny4kOLED - Drivers for SSD1306 controlled dot matrix OLED/PLED 128x32 displays
 *
 * Based on ssd1306xled, re-written and extended by Stephen Denne
 * from 2017-04-25 at https://github.com/datacute/Tiny4kOLED
 *
 */

// Choose your I2C implementation before including Tiny4kOLED.h
// The default is selected is Wire.h

// To use the Wire library:
// This example compiles to 4402 bytes of program storage space
// and 88 bytes of dynamic memory.
//#include <Wire.h>

// To use the Adafruit's TinyWireM library:
// (Saves about 350 bytes and 20 bytes of RAM over Wire.h)
// (If you see a strange dot pattern then upgrade the TinyWireM
//  library to get the buffer overflow fix.)
//#include <TinyWireM.h>

// To use the TinyI2C library from https://github.com/technoblogy/tiny-i2c
// (Saves about 570 bytes and 40 bytes of RAM over Wire.h)
//#include <TinyI2CMaster.h>

// The blue OLED screen requires a long initialization on power on.
// The code to wait for it to be ready uses 20 bytes of program storage space
// If you are using a white OLED, this can be reclaimed by uncommenting
// the following line (before including Tiny4kOLED.h):
//#define TINY4KOLED_QUICK_BEGIN

#include <Tiny4kOLED.h>

// ============================================================================

// This example simply shows the number of milliseconds since reset
// To show the double-buffering effect, the location where the timer
// is displayed moves left and right, and the screen is completely cleared.

// The number increases, so if the location was not moving,
// we would not need to repeatedly clear and initialize the display.
// But we are doing so to demonstrate the capability of the double
// buffering code to result in a smooth animation

// Uncomment this line to disable the code performing the double buffering
//#define NO_DOUBLE_BUFFERING

void setup() {
    oled.begin(sizeof(tiny4koled_init_128x64r), tiny4koled_init_128x64r);

  oled.setFont(FONT6X8);
  updateDisplay();
#ifndef NO_DOUBLE_BUFFERING
  oled.switchRenderFrame();
  updateDisplay();
  oled.switchFrame();
#endif
  oled.on();
}

void loop() {
  delay(50);
  updateDisplay();
#ifndef NO_DOUBLE_BUFFERING
  oled.switchFrame();
#endif
}

void updateDisplay() {
  oled.clear();

  oled.setCursor(0, 0);
  oled.println(F("i am Xnork"));
  oled.println(F("i am Xnork"));
  oled.println(F("i am Xnork"));
  oled.println(F("i am Xnork"));
  oled.println(F("i am Xnork"));
  oled.println(F("i am Xnork"));
  oled.println(F("i am Xnork"));
  oled.println(F("i am Xnork"));
  oled.println(F("i am Xnork"));
  oled.println(F("i am Xnork"));
  oled.println(F("i am Xnork"));
  oled.println(F("i am Xnork"));
  oled.println(F("i am Xnork"));
  oled.println(F("i am Xnork"));
  oled.println(F("i am Xnork"));
}
```

When i upload this code to MH Attiny88, show but with flicking... Only when i change default size and font size....

datacute, 2023-09-24:
Do you expect the screen to scroll? After a few lines, you keep overwriting the same text at the bottom.

Xnork, 2023-09-24:
When I change the screen size, to 128x64 it starts showing the text 2 times below.

I did some tests and it seems to be when rendering the pages, no matter how much I readjust the height, it gets buggy on the 4th line down.

[image]

```c++
#include <Tiny4kOLED.h>

void setup() {
  oled.begin(sizeof(tiny4koled_init_128x64br), tiny4koled_init_128x64br);
  oled.setFont(FONT6X8);

  oled.clear();
  updateDisplay();
  oled.switchRenderFrame();
  
  oled.on();
}

void loop() {
  updateDisplay();
  oled.switchFrame();
  oled.clear();
}

void updateDisplay() {
  oled.clear();
  oled.setCursor(0, 0);
  oled.print(F("millis: "));
  oled.println(millis());
  oled.println(F("/root $ "));
}
```

datacute, 2023-09-24:
Use the begin method that takes with and height too.

Xnork, 2023-09-25:
Like that?
```c++
oled.begin(128, 64, sizeof(tiny4koled_init_128x64br), tiny4koled_init_128x64br);
```

I already tried that, nothing changes.

datacute, 2023-09-25:
Yes, and when using the 128x64 size display, you don't have 2 frames to use, so you need to remove the  oled.switchRenderFrame(); and oled.switchFrame();
With the 128x32 display you can write to memory that is 'offscreen', then switch to it. With the 128x64 display you cannot, as all its memory is visible.
The flickering is due to calling oled.clear(). An alternative strategy is to just update portions of the screen that change - for this example, just writing the millis.

closed 2023-09-27 by Xnork