datacute/Tiny4kOLED #58: Initializing DCfont with local uint8_t* array leads to no text being drawn on screen

HyperLethalVector, 2023-10-12:
Hoii!

I've been trying to load in a DCfont dynamically from a series of stored fonts in an SPI flash module.

The bytes are correctly loaded, however allocating the locally created font causes blank text on the screen.

To replicate:

- Load any font (I used FONT5X5), draw text on screen
- Re-assign the bitmap variable to another uint8_t* OR create a local variable 

Code sample below:

```

DCfont* f;
void TestFonts(
MemoryAddresses& configOut){
  uint8_t buffer[configOut.FontOneInfo.SizeF1];  
  Serial.println(F("===FONT==="));  
  int j = 0;
  for(int i = 0; i < configOut.FontOneInfo.SizeF1; i++){
      uint8_t b2 = flash.readByte(configOut.FontOneInfo.AddressF1+i);
      buffer[i] = b2;
      char s[7];
      sprintf(s, "0x%02x", b2);
      j++;
      if(j > 7){j = 0; Serial.print(s); Serial.println(","); }
      else{ Serial.print(s); Serial.print(",");}
  }  
  f = new DCfont{};
  f->spacing = TinyOLED4kfontPapyrus.spacing; // this is my alternative naming for the papyrus font, used for texting
  f->bitmap = buffer;// this is another pre-defined font based on the papyrus text
  f->width = TinyOLED4kfontPapyrus.spacing;
  f->height = TinyOLED4kfontPapyrus.height;
  f->widths = TinyOLED4kfontPapyrus.widths;
  f->widths16s = TinyOLED4kfontPapyrus.widths16s;
  f->spacing = TinyOLED4kfontPapyrus.spacing;
  f->first = TinyOLED4kfontPapyrus.first;

  oled.off();
  oled.clear();
  oled.setFont(FONT5X5);  //This works fine
  oled.setCursor(0,0);
  oled.println("Do I look Pretty?");    
  oled.on();  
  Serial.println(F("==Done=="));  
  delay(2000);
  oled.off();
  oled.clear();
  
  oled.begin(sizeof(ssd1306_init_sequence4),ssd1306_init_sequence4);
  oled.off();
  oled.setFont(f); // This does not work
  oled.setCursor(0,1);
  oled.println("Testing me now");
  oled.on();  
  Serial.println(F("==Done=="));  
  delay(2000);  
}
```

datacute, 2023-10-12:
The bitmap data is expected to be a reference to PGM space, and is read with the pgm_read_byte macro.

HyperLethalVector, 2023-10-12:
that makes alot of sense, guess I'm going to have to do a little bit of engineering for it to achieve my goals. 

I'll close this ^__^

closed 2023-10-12 by HyperLethalVector