datacute/Tiny4kOLED #59: 0.96" vs 1.3" oled

sizzlik, 2023-11-10:
Hi..i need a little help to get a 1.3" oled to work properly with an attiny85 (dip 8). I got a small one, think its .96" 128x64. I dont know where i got it from, maybe an arduino sensor/starter kit so i dont have a datasheet but it works fine with the library. Then i got 1.3" oleds from amazon (https://www.amazon.de/gp/product/B07FYG8MZN/) OLED Display I2C SSH1106 Chip 128 x 64..and on those everything seems zoomed in, just every 2nd line is used, doesnt scroll etc. Im too dumb to find my error. Is the display chipset maybe not supported? Do i have to change a setting in header files? I spend all night trying to find a easy to use library to display text from an attiny85 on an oled..this one was the first to work.
I would be endless thankfull if someone could help me out how to display text on the 1.3" oled via tiny85 with as less code as possible. All i need is something like the lcd drivers: oled.clear(), oled.setCursor(x, y), oled.print("it works")..no external input, no graphics..just setting the curosr into position, write text, clear and repeat.
Thanks for any help!

datacute, 2023-11-10:
While this library is designed for the SSD1306 and the SSH1106 is obviously a different chip, you can probably get most of the functionality working just with variations of the `oled.begin()` command.

The biggest difference is that the SSH1106 supports a width of 132 pixels instead of 128 pixels.

You will need to use the `begin(uint8_t xOffset, uint8_t yOffset, uint8_t width, uint8_t height, uint8_t init_sequence_length, const uint8_t init_sequence [])` overload of `begin` since otherwise the xOffset defaults to half of (128 - width).

Examples of different `init_sequence` commands to send can be found in https://github.com/datacute/Tiny4kOLED/blob/master/src/Tiny4kOLED_common.h#L231

Code examples using these initialisation sequences can be found in the Examples/Devices folders, for various sized displays. 

There is no SSH1106 example or init sequence in this library though.

The use of every second line is due to variations in how different screens using the SSD1306 (or SSH1106) chip arrange the mapping of memory to lines of pixels. You can use the `oled.setComPinsHardwareConfiguration` method to try variations, but it is probably due to you simply using `oled.begin()` which defaults to using initialisation commands `0xDA, 0x12` appropriate for most 128x32 screens (the initial target for this library, which I've kept for backwards compatibility reasons). Most 128x64 screens  need the command `0xDA, 0x02` instead. This can be supplied in the init sequence used by the begin call, or by calling `setComPinsHardwareConfiguration` afterwards.

Note: Every 1.3" screen I've seen has been an SSH1106. 0.96" is the largest I've seen using the SSD1306.

datacute, 2023-11-10:
TL;DR answer: `oled.setComPinsHardwareConfiguration(0,1);`

datacute, 2023-11-10:
TL;DR answer: 
try `oled.setComPinsHardwareConfiguration(1,1);`
or `oled.setComPinsHardwareConfiguration(0,1);`

sizzlik, 2023-11-11:
thanks for the extensive answer. Sadly i cant get it to display the text properly. Guess i have to keep on googling my way to a quick sollution/other library or worst case use the small '96 oled or an 20x4 lcd screen. But thank you very much for taking the time to answer, much appreciated.

sizzlik, 2023-11-11:
closed as user too dumb =)

closed 2023-11-11 by sizzlik