datacute/Tiny4kOLED #62: SSD1306 is displaying nothing

Martius108, 2024-05-13:
Hello,

probably I'm doing something wrong or I didn't understand the usage of this library properly. I have a SSD1306 128x64 display. My ATtiny85 has a Bootloader set to 8 MHz. I use the ATtinyCore from Spencer Conde with the Optiboot option to program the Attiny. When I upload one of your test sketches (from the Github site here or from the examples in the Arduino IDE) there is no output on the display. I normally use the TinyWireM library from Adafruit.

datacute, 2024-05-13:
It could be that the SSD1306 has remembered some settings stopping it from working> This Tiny4kOLED library uses a minimal setup routine, which assumes the SSD1306 starts with its default values. If this is the case, you could try installing and running the "ResetToDefaults" example?

Martius108, 2024-05-13:
No, unfortunately it isn't showing nothing is this case as well. But the display is ok, I have another ATtiny project laying around here and as soon as I put it in here it shows the expected values. Here is the link to the code: https://github.com/SensorsIot/BatteryTesterAttiny

Martius108, 2024-05-13:
Ok, solved. It had something to do with my self built programmer :)

datacute, 2024-05-13:
:)

closed 2024-05-13 by datacute