datacute/Tiny4kOLED #63: How to clear certain lines/values without oled.clear()?

Martius108, 2024-05-31:
Hello,

I would like to do something like this (this is not valid code; just to give you an idea):

void setup():
oled.setCursor(0, 0);
oled.println(„Headline“);

void loop():
oled.setCursor(0, 8);
oled.print(time);
oled.println(„ ms“);

How can I achieve that the headline is always on top without refreshing itself during every loop and only the time values are changing? Normally I would work with oled.clear() but this clears the headline too. In the Adafruit SSD1306 library this is somehow easier, kind of built-in, the values are overwritten but the old pixels aren’t show up.
In the Tiny4k library I see the old pixels when I skip the oled.clear() function.

datacute, 2024-05-31:
Most likely the number of pixels to show the time is changing, resulting in pixels on the right not being overwritten?
If this is what is happening, you can use
oled.print(" ms");
oled.clearToEOL();

If you know the maximum number of columns of pixels you need to clear, and it is a 1 page high font, you can call oled.fillLength(length);

Each of these leave the cursor on the same line, so you are wanting to print more text on the next line you would need to call println() or setCursor().

Martius108, 2024-06-01:
Thanks.
I have one page and one line as headline followed by either 3 lines (all text with double height; so 8 standard lines in total) or one empty line followed by 2 lines with text and changing data. The whole thing is a tachometer and I display either rpm, distance and speed or just distance and speed. The headline is just „Tachometer“, which appears yellow because a have this two color SSD 1306 display.
So which method would you recommend?

Martius108, 2024-06-02:
I also tried it with an extra function but this is not working either ..

`// Tachometer ATtiny85

#include <util/atomic.h>
#include <TinyWireM.h>
#include <Tiny4kOLED.h>

const int hallPin = PB4;           // Digital pin hall sensor
int lastHallState = 0;             // Last hall state
unsigned long lastTime = 0;        // Last time
int revolutions = 0;               // Revolutions
float distance = 0.0;              // Distance
float totalTime = 0.0;             // Total time
const float revDistance = 0.00223; // Distance per revolution; 28" in km (223 cm = 0.00223 km)
float revSpeed = 0.0;              // Speed in km/h

void setup() {
  pinMode(hallPin, INPUT_PULLUP);  // Pullup is needed here

  oled.begin(128, 64, sizeof(tiny4koled_init_defaults), tiny4koled_init_defaults);   // Init display
  oled.enableChargePump();    // The default is off, but most boards need this.
  oled.setRotation(1);        // The default orientation is not the most commonly used.
  oled.clear();               // Clear display
  oled.setFontX2(FONT6X8P);   // Set fonts
  oled.on();
}

void loop() {
  // Show the previous values until new trigger impulse
  oled.setCursor(0, 0);       // Set cursor
  oled.println("Tachometer"); // Headline
  oled.print(distance, 1);
  oled.println(" km");
  oled.print(revSpeed, 1);
  oled.println(" km/h");
  
  int currentHallState = digitalRead(hallPin);

  if (currentHallState != lastHallState) {  // Check if status has been changed
    
    if (currentHallState == LOW) {          // Check if sensor state is low
      revolutions++;                
      distance = revolutions * revDistance; // Make all necessary calculations
      float revTime = (millis() - lastTime) / 1000.0 / 3600.0;
      totalTime = totalTime + revTime;
      revSpeed = revDistance / revTime;

      deleteDataDisplay(2);
      deleteDataDisplay(4);
      
      if(distance >= 10.0) {
        oled.print(distance, 1);
      } else {
        oled.print(distance, 2);
      }                     
      oled.println(" km");
      oled.print(revSpeed, 1);
      oled.println(" km/h");
   
      lastTime = millis();                  // Set lastTime to be the new value
    } 
  }  
  lastHallState = currentHallState;         // Set the lastHallState to be the new one
}

void deleteDataDisplay(byte line) {
  oled.startData();
  oled.setCursor(0, line);
  oled.repeatData(0x00, 61);
  oled.endData();
  oled.setCursor(0, line + 1);
  oled.startData();
  oled.repeatData(0x00, 61);
  oled.endData();
}`