datacute/Tiny4kOLED #67: Digispark ATiny85

Silvanskii, 2024-11-19:
I tried some of your examples and they don't display correctly.
[image]
[image]

datacute, 2024-11-19:
I can think of two possibilities:
1. The screen has some non-default settings that it is remembering. My library doesn't set all configuration settings, only non-default ones. Running the ResetToDefaults example would fix that.
2. The communication to the device is not working properly. I think this is more likely, because your 2 screen shots both show missing pixels, and the other strange effects could also be caused by data being corrupted as it is sent to the screen. I'm not sure how you should solve this though. Pull ups? Choice of wire library and i2c speed? Changing to a different cable?

Silvanskii, 2024-11-19:
when i use [TinyOzOled](https://github.com/SensorsIot/TinyOzOled)
[image]
when i use SSD1306Ascii
[image]
It's the cable, specifically the library. When I use the DigisparkOLED library I see in the demo a white stripe of 1-2 pixels, which is not removed. I tried another display of the same type, the situation is the same.  I soldered resistors, the situation is the same. So far the SSD1306Ascii library works best, but I need to show 1 picture, not text